hannobraun/stepper #3: Add support for EN/FAULT pin

hannobraun, 2020-10-19:
The driver API does not have support for controlling the power stage or reading a fault. There's a bit of a difficulty here, since the same pin is used for both, and I don't believe there are traits in embedded-hal that could be used for both setting output and reading input on the same pin.

jessebraham, 2020-11-08:
To further complicate things, drivers such as the [DRV8825](https://www.pololu.com/product/2133) have separate `Enable` and `Fault` lines.

WriterOfAlicrow, 2021-07-30:
> The driver API does not have support for controlling the power stage or reading a fault. There's a bit of a difficulty here, since the same pin is used for both, and I don't believe there are traits in embedded-hal that could be used for both setting output and reading input on the same pin.

Looks like embedded-hal supports it now. They added `IoPin` which can toggle between input and output. See the bottom of https://github.com/rust-embedded/embedded-hal/blob/master/src/blocking/digital.rs

hannobraun, 2021-07-30:
Nice, thanks for the note, @WriterOfAlicrow!