hannobraun/stepper #14: `StepMode` assumes 4 configuration inputs

jessebraham, 2020-10-21:
First off I just wanted to say thanks for your work on this project, I'm looking forward to using it!

I have a [Pololu DRV8825](https://www.pololu.com/category/154/drv8825-stepper-motor-driver-carriers-high-current) board kicking around so I figured I'd take a swing at writing a driver for it using `step_dir`. It's been relatively painless to adapt the reference implementation to this board so far, however I did encounter an issue with setting the step mode.

Currently the library has a pair of `StepMode` implementations:
- one with the function [`to_signals`](https://github.com/braun-embedded/step-dir/issues/new) 
- a [`TryFrom<u16>`](https://github.com/braun-embedded/step-dir/blob/master/src/lib.rs#L133-L146) implementation.

These functions make the assumption that there are 4 inputs for step mode configuration, however other boards use 3 or even 2 inputs instead. I just wanted to bring this to your attention as well was ask/discuss what the best path forward is.

hannobraun, 2020-10-21:
Thank you, good catch! I moved those over from the stspin220 code yesterday, and I missed those implementation-specific parts.

I'm very glad to get some scrutiny this early! I expected to work on this for a few months more (adding PWM/DMA-backed interfaces, more drivers) before someone became interested :-)

Before I get into your points, for the sake of fairness I have to mention that there [already is a DRV8825 driver](https://crates.io/crates/stepper-driver). I decided to ignore that and build my own stuff, because I want to go much farther (the aforementioned PWM/DMA-backed interfaces), and there doesn't seem to be any ongoing development there.

Now, to your points:
- `to_signals` is a clear oversight. I'd say, convert it into a free function and move it to `stspin220`. It might make sense to move it somewhere else later, to share code between drivers, but that doesn't need to concern us now.
- I don't see a direct problem with `TryFrom<u16>`. The only problem I see is with `StepMode`/`SetStepMode` overall, in that they currently don't offer anything to drivers that don't support the full 1/256th microstepping resolution.

I haven't put much thought into how to support drivers with different microstep resolutions so far. I see two ways to do that:

1. Going all static. Add an associated type to `SetStepMode` that defines the microstep enum (we can generate all possible enums in a macro, then each driver wouldn't have to define their own). Add appropriate `TryFrom`/`Into` bounds to that associated type, for conversions from/to numbers.
2. Keep the current enum. Add an associated constant to `SetStepMode` to define what the maximum resolution for a given timer is. Expect users to check this constant, and throw runtime errors if they call `set_step_mode` with an invalid value.

Now that I've written it out, 1. seems to be the better solution to me.

---

If you'd consider contributing your driver to this crate, that would be a very welcome addition! I intend this to be the one-stop shop for any stepper motor drivers that are controlled with STEP and DIR signals, so more drivers are very much in scope. That would also help keep the driver up-to-date, as more features are added to the generic interfaces.

If you want to implement any of the improvements I suggested in this comment (or suggest your own), that would also be very welcome. If not, I will make some time to do it myself soon-ish. (I actually also intend to add drivers for most of the [driver chips available on Pololu boards](https://www.pololu.com/category/120/stepper-motor-drivers), to get things going here, but that will take some time. Right now I'm focused on creating a test stand to get some automated testing going, and after that I'll likely work on a PWM-based interface.)

jessebraham, 2020-10-21:
> `to_signals` is a clear oversight. I'd say, convert it into a free function and move it to stspin220. It might make sense to move it somewhere else later, to share code between drivers, but that doesn't need to concern us now.

This sounds good to me, I agree trying to share this code is probably premature at this point in time. I will implement a similar function in my driver crate for now.

> I don't see a direct problem with `TryFrom<u16>`.

My mistake here, I don't know what I thought it was doing 😁 After looking at it with fresh eyes I agree it's fine.

> The only problem I see is with `StepMode`/`SetStepMode` overall, in that they currently don't offer anything to drivers that don't support the full 1/256th microstepping resolution.

Yeah that is a bit of an issue. The DRV8825 only goes down to 1/32 steps, so there's an incompatibility there already. I agree that going the associated type route for this is the better of the two choices you proposed. Seems like a reasonable thing to be defined at the driver level.

>If you'd consider contributing your driver to this crate, that would be a very welcome addition! I intend this to be the one-stop shop for any stepper motor drivers that are controlled with STEP and DIR signals, so more drivers are very much in scope.

That was my intention from the start! Once I've finished writing, testing, and documenting the driver I will definitely open a PR. I'll probably even grab some more boards eventually and write drivers for them too, but one thing at a time 😉 

> If you want to implement any of the improvements I suggested in this comment (or suggest your own), that would also be very welcome.

Can't make any promises but I'll do what I can! I'll spend some more time thinking about how we can improve the library at the very least.

Thanks for the feedback!

hannobraun, 2020-10-21:
> That was my intention from the start! Once I've finished writing, testing, and documenting the driver I will definitely open a PR. I'll probably even grab some more boards eventually and write drivers for them too, but one thing at a time :wink:

I love this! :smile: 

> Can't make any promises but I'll do what I can! I definitely have intentions of doing so. I'll spend some more time thinking about how we can improve the library at the very least.

No worries. I'll happily make those improvements myself (probably next week or so), but you gotta give those potential contributors a chance to rope themselves in. :grin:

> Thanks for the feedback!

My pleasure. Thanks again for opening the issue, and your interest in the project.

PR #15 (2020-11-03): Support drivers with other microstepping resolutions
Extends `SetStepMode` with a `StepMode` associated type, so drivers can choose the right type to represent their capability. Renames the current `StepMode` enum to `StepMode256` and adds a bunch of other enums to represent other microstepping resolutions.

See the commits themselves for more details.

Close #14
closed 2020-11-03 by hannobraun