hannobraun/stepper #18: Consider adding vendor documentation to repository

hannobraun, 2020-11-11:
I think it would be nice to have a more principled approach to vendor documentation than just sprinkling links in the comments. We could add PDFs to a `documentation/` directory, or we could just add a Reference section to the README and link vendor documentation there (probably best to do so via archive.org).

Not sure which approach I prefer.

hannobraun, 2021-02-03:
Addendum: I've heard some rumors about TI causing trouble regarding copyright violations of their documentation. If true, this is bullshit of course (and it's bullshit to apply an "all rights reserved" copyright to freely available documentation anyway), but I think we shouldn't put any copyrighted material into this repository, no matter how much sense it makes.

Referencing links in README is probably the way to go here.

jessebraham, 2021-02-11:
I've been hearing similar rumors as well in recent weeks, best not to play with fire.

commit 46fee260 (2021-04-30): Link vendor documentation for each driver

Close #18
closed 2021-04-30 by hannobraun