hannobraun/stepper #39: Document use of `generate-drivers` task

hannobraun, 2020-12-02:
There is currently no documentation for the `generate-drivers` task. The following aspects should be documented:

- That drivers are generated from a template and shouldn't be modified directly.
- How to generate the drivers.

The bulk of that should go into a new section in `CONTRIBUTING.md` (maybe "how to add a driver"?). The release procedure section also needs to be updated. In addition, it would be nice to have a `README.md` in `drivers/`, that notes that drivers shouldn't be modified directly and points to `CONTRIBUTING.md`.

---

Here's how to generate the drivers:
```
$ cargo install cargo-task
$ cargo task generate-drivers
```

FFY00, 2021-03-20:
Hey, could you write some information on this? I am looking to add support for the TMC2209 but I am not sure how to proceed.

hannobraun, 2021-03-20:
Hey, @FFY00. Thanks for your interest! Support for more drivers is definitely welcome (I just haven't gotten to it yet; so much to do), so I'll help you as best I can. And by the way, writing some high-level documentation (including how to write new drivers) is going to be my next focus, but I'm on vacation right now, and won't have a lot of time starting April, so it may take a while.

I've [written up some information](https://github.com/flott-motion/stepper/issues/72#issuecomment-803321380) in #72, as that's the more topical issue. Please let me know if anything is unclear.

FFY00, 2021-04-08:
Hi, thank you for your quick reply, and sorry for my late one. I am also a bit busy myself, it's hard to manage life, oss, and other responsibilities. I am definitely interested in moving this forward, though the project I need this for is not really a big priority right now. I will take a look and try to progress as time allows.

hannobraun, 2021-04-08:
Totally understood. Let me know, if you have any more questions, happy to help.

commit 0ad3d8f4 (2021-04-30): Document how to create driver crates

Close #39
closed 2021-04-30 by hannobraun