hannobraun/stepper #44: Add test suite for DRV8825

hannobraun, 2020-12-07:
We now have a test stand in the repository that runs a test suite for the STSPIN220. It would be great to also have a test suite for the DRV8825. I think for now it would be good enough to just add another test suite (modeled after the STSPIN220 test suite) and tell users to pass `-- test drv8825` or `--test stspin220`, depending on what they have available.

hannobraun, 2020-12-10:
Largely blocked by #49.

jessebraham, 2020-12-12:
I'm able to tackle this over the next few days if you feel it's still worth pursuing at this point. It's fairly derivative of the `STSPIN220` tests, so it shouldn't be too much effort.

hannobraun, 2020-12-12:
I'd say it's worth pursuing if you're interested in the test stand and want to look into it at this point (as a learning experience, or to figure out how to make it work better, for example). If you're not interested in that, your time is probably better spent elsewhere, until someone builds a better test stand.

hannobraun, 2024-01-23:
This library is no longer actively developed (and hasn't been for a while). I'm closing any issues that don't relate to specific gaps in existing functionality, as to not make the wrong impression about what is planned and desired.

closed 2024-01-23 by hannobraun