hannobraun/stepper #49: Test stand is too unreliable for practical use

hannobraun, 2020-12-10:
In #26, I had gotten the test stand to a pretty reliable state. However, as it turns out, that didn't last. When I got it out to test something else later (I don't have the room on my desk to leave it set up permanently), the test stand was very unreliable. It miscounted steps and reported and panicked due to an unexpected direction being detected.

The current hardware setup is definitely too flimsy. The encoder needs to be mounted more securely, possibly a better encoder needs to be used, and the wiring should be more solid (ideally, we'd have a custom PCB). There might also be problems on the software side, but as long as the hardware side is so unreliable, I'm in no mood for hunting down such potential problems.

I think the test stand served its purpose as a first prototype, but it's not practical enough going forward. I think we need an improved version that incorporates the experiences from the current one. I intend to look into that, but for the time being, I'll be busy with other things. I think for now, we should keep its code up-to-date, to make an eventual second version easier, but I don't think we should rely on it otherwise.

jessebraham, 2021-02-11:
While browsing for parts for another project, I've stumbled upon what I hope is a viable solution for the test stand. I have ordered the required components and am just waiting on shipping (so it could be a couple weeks, potentially).

I picked up a some optical incremental encoders and encoder wheels, both having resolutions of 200 counts per rotation. In theory each step (in full-stepping mode) will generate a corresponding encoder signal with this setup (rather than the 10:1 ratio we had before). Higher resolution encoders and wheels are available as well, so if this proves to be viable we could potentially test micro-stepping modes too.

Once the parts have arrived and I've had a chance to play around with the setup I'll let you know the results.

hannobraun, 2021-02-11:
Nice, looking forward to it! I've also been looking into better encoders (although I didn't spent a lot of time) and haven't come up with anything good so far.

commit a4fa6bde (2021-12-13): Remove test stand

It has never been working correctly (see #49), and just having it here
causes quite a bit of maintenance effort.

Close #49
PR #132 (2021-12-13): Remove test stand
It has never been working correctly (see #49), and just having it here
causes quite a bit of maintenance effort.

Close #49
closed 2021-12-13 by hannobraun