hannobraun/stepper #147: Expose a StepMode's maximum step denominator as a const

shyndman, 2022-11-28:
Hi there,

I'm not sure this would be useful generally for the library, but in my fork I've made a change to `StepMode` so that it exposes a `MAX_STEP_BASE` constant. eg:

```rust
impl StepMode for StepMode16 {
    const MAX_STEP_BASE: u16 = 16;
}
```

My application code is driver-agnostic, and as such, I've been using this constant in a bunch of different ways.

```rust
// Setting the base of a rational number representing the driver's current step:
let step = Rational32::new_raw(1, Driver::StepMode::MAX_STEP_BASE as i32);

// Getting the base without having to iterate over the modes:
let max_base: u16 = Driver::StepMode::MAX_STEP_BASE.try_into()?;
```

What do you think? Useful?

shyndman, 2022-11-28:
And for the record, I'm not crazy about the naming. If you're into the idea, I'm interested in hearing suggestions.

hannobraun, 2022-11-29:
I haven't actively used this library in a long while, so I'm not really in a position to judge this. It certainly sounds useful, and obviously it's useful for you. That's good enough, in my opinion. Happy to accept such a pull request!

closed 2024-01-23 by hannobraun