hannobraun/stepper #149: Issues with fugit_timer::Timer traits

sadilekivan, 2022-12-12:
I'm still fairy new to rust and trying to make this crate work with a `step/dir` motor on a Raspberry Pi. For that I'm utilizing the `linux_embedded_hal`. I made most of the traits work, but I'm still missing the timer implementation and I am confused to as how to do that.

There is a `linux_embedded_hal::SysTimer` that is missing the `fugit_timer::Timer`. Do I implement that, or am I in the wrong place, and need some other timer?

hannobraun, 2022-12-12:
Hey @sadilekivan! I haven't used this crate in a long time, nor have I done any embedded development recently in general, so unfortunately I'm not sure what the solution is.

I've also never used `linux-embedded-hal`, but based on my experience with other HALs, I think it might be reasonable to implement the `fugit` traits there. It looks like their `SysTimer` type implements `embedded-hal`'s `CountDown`, which has been removed in the latest alpha versions. It might be appropriate for them to support a different timer trait from a third-party library like `fugit`. This is probably a question that should be directed to the `linux-embedded-hal` maintainers. 

In the meantime, you can probably work around that, although you can't implement the `fugit` trait directly. When implementing a trait in Rust, either the trait or the type you implement it for need to be in your local crate. What you can do though, is define your own timer type and implement the `fugit` trait for that.

In your situation, it might make sense to build that timer type directly on the `Instant`/`Duration` types from the standard library, bypassing `linux-embedded-hal`. The documentation of `linux_embedded_hal::SysTimer` says that it's implemented on top of `std::time::Instant` anyway, and it looks like `SysTimer` was specifically created as an implementation of `CountDown`. It would probably be a pain to use it as a basis for your own implementation of `fugit`'s `Timer` trait.

hannobraun, 2024-01-23:
This library is no longer actively developed (and hasn't been for a while). I'm closing any issues that don't relate to specific gaps in existing functionality, as to not make the wrong impression about what is planned and desired.

closed 2024-01-23 by hannobraun